raspberrypi/usbboot #176: Build usbboot on macOS with apple silicon

ApiTracer, 2022-12-16:
### Describe the bug

On Apple silicon Mac, there is an error on make command :
ld: warning: ignoring file /usr/local/Cellar/libusb/1.0.26/lib/libusb-1.0.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
....

I applied this workaround : 
Download the libusb source from github.com:
https://github.com/libusb/libusb/releases/tag/v1.0.26

Build it in the standard way:
extract libusb-1.0.26.tar.bz2
$ cd libusb-1.0.26
$ ./configure
$ make
$ make check
$ sudo make install

and make again without errors.

### Steps to reproduce the behaviour

Try to build on macOS with apple silicon

### Device(s)

Other

timg236, 2022-12-16:
There's no official support for macOS, but happy to leave the bug open in-case anyone has any suggestions.

pelwell, 2022-12-16:
Out of interest, does modifying the Makefile to say `--libs libusb` (instead of `--libs libusb-1.0`) make any difference?

ApiTracer, 2022-12-16:
> Out of interest, does modifying the Makefile to say `--libs libusb` (instead of `--libs libusb-1.0`) make any difference?

usbboot % make
cc -Wall -Wextra -g -o rpiboot main.c `pkg-config --cflags --libs libusb` -DGIT_VER="\"cdf79ac5\"" -DPKG_VER="\"20221104~153421\""
Package libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb' found
main.c:1:10: fatal error: 'libusb.h' file not found
#include <libusb.h>

It seems to be not found with this notation

timg236, 2022-12-16:
Did you try this workaround from Readme.md

`If the build is unable to find the header file libusb.h then most likely the PKG_CONFIG_PATH is not set properly. This should be set via export PKG_CONFIG_PATH="$(brew --prefix libusb)/lib/pkgconfig".`

ApiTracer, 2022-12-16:
same error after launch the export, with libusb-1.0, the package is found but with the wrong arch

pelwell, 2022-12-16:
I think this can become a note in the README.

timg236, 2023-05-11:
Seems to be resolved by commits mentioned above.

closed 2023-05-11 by timg236