raspberrypi/usbboot #177: Unable to load rpiboot on Macbook pro with RPI CM4 + Home Assistant Yellow

avermeer-tc, 2022-12-19:
### Describe the bug

I have a MacBook Pro directly connected via a USB C cable to the Yellow. I've removed all peripherals from the Yellow board.

```
usbboot git:(master) ✗ sudo ./rpiboot
RPIBOOT: build-date Dec 19 2022 version 20221215~105525 e4819de4
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Second stage boot server
Cannot open file config.txt
Cannot open file pieeprom.sig
Loading embedded: start4.elf
File read: start4.elf
Cannot open file fixup4.dat
Second stage boot server done```

Nothing happens. The green light does pop on. But no storage device is detected. I tried via Windows as well and I get the same error.

```
sudo ./rpiboot -d mass-storage-gadget
RPIBOOT: build-date Dec 19 2022 version 20221215~105525 e4819de4
Loading: mass-storage-gadget/bootcode4.bin
Waiting for BCM2835/6/7/2711...
Loading: mass-storage-gadget/bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading: mass-storage-gadget/bootcode4.bin
Second stage boot server
Loading: mass-storage-gadget/config.txt
File read: config.txt
Loading: mass-storage-gadget/boot.img
File read: boot.img
Second stage boot server done
```

Still no disk is detected, no green LED light.

### Steps to reproduce the behaviour

Connect the USB directly to the Yellow board. Jumpers set for USB + USB-C Rcvry. Run rpiboot. See output from above.

### Device(s)

Raspberry Pi CM4

### Compute Module IO board.

Home asstant Yellow rev 1.2

### RPIBOOT logs

sudo ./rpiboot
RPIBOOT: build-date Dec 19 2022 version 20221215~105525 e4819de4
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Second stage boot server
Cannot open file config.txt
Cannot open file pieeprom.sig
Loading embedded: start4.elf
File read: start4.elf
Cannot open file fixup4.dat
Second stage boot server done

### Kernel logs

➜  usbboot git:(master) ✗ sudo dmesg | grep -i usb
[  816.480831]: 000816.480829 HS02@14200000: AppleUSBHostPort::createDevice: failed to create device (0xe00002bc)
[  818.811973]: 000818.811971 HS02@14200000: AppleUSBHostPort::createDevice: failed to create device (0xe00002bc)
[  821.207744]: 000821.207742 HS02@14200000: AppleUSBHostPort::createDevice: failed to create device (0xe00002bc)
[  823.670892]: 000823.670889 HS02@14200000: AppleUSBHostPort::createDevice: failed to create device (0xe00002bc)
[  826.197160]: 000826.197157 HS02@14200000: AppleUSBHostPort::createDevice: failed to create device (0xe00002bc)
[  826.197189]: 000826.197188 HS02@14200000: AppleUSBHostPort::disconnect: persistent enumeration failures

timg236, 2022-12-19:
The logs suggest that rpiboot is loading the software successfully but MacOS ignores the MSD descriptors. You could confirm that by connecting to the CM4 in mass-storage-gadget mode via the UART. 
We don’t test on MacOS but if anyone can see a problem in the mass storage gadget descriptors that causes MacOS to ignore them then we can review that.

avermeer-tc, 2022-12-21:
I am to blame here. I forgot that I have a CM4-Lite. So there is no eMMC to mount. My bad.

closed 2022-12-21 by avermeer-tc