raspberrypi/usbboot #183: make not working on Ubuntu 22.04

rebeccaRossRobotics, 2023-05-19:
### Describe the bug

I followed the instructions for building on linux, but get an error when running `make`

```
rebecca@rebecca-ThinkPad-L14-Gen-1:~$ cd usbboot/
rebecca@rebecca-ThinkPad-L14-Gen-1:~/usbboot$ make
cc -Wall -Wextra -g -o bin2c bin2c.c
make: cc: No such file or directory
make: *** [Makefile:13: bin2c] Error 127
rebecca@rebecca-ThinkPad-L14-Gen-1:~/usbboot$ 
```

### Steps to reproduce the behaviour

- Be on a Computer with Ubuntu22.04 installed.
- sudo apt install git libusb-1.0-0-dev pkg-config
- git clone --depth=1 https://github.com/raspberrypi/usbboot
- cd usbboot
- make

### Device(s)

Other

pelwell, 2023-05-19:
The error message is saying that "cc" isn't a recognised program. Either you have no C compiler installed or "make" hasn't been configured to use it by default.

What does `which gcc` report? If it finds something, try `make CC=gcc`.

rebeccaRossRobotics, 2023-05-19:
Hi Thanks for the speedy response!

`which gcc` doesn't return anything. 
```
rebecca@rebecca-ThinkPad-L14-Gen-1:~$ which gcc
rebecca@rebecca-ThinkPad-L14-Gen-1:~$ 
```

I assume that means i haven't got a c compiler installed as you suggested. Which would make sense as I clean installed ubuntu a couple of days ago

ghollingworth, 2023-05-19:
```sudo apt install build-essential```

Should be the right thing then.

If this works, please close the issue... Thx

rebeccaRossRobotics, 2023-05-19:
yes that was it! thanks so much!

closed 2023-05-19 by rebeccaRossRobotics