raspberrypi/usbboot #187: CM4 not being recognized

Roycinger, 2023-09-15:
### Describe the bug

After starting either rpiboot.exe or rpi-mass-storage-gadget.bat, the CM4 is not being recognized and gets stuck, see screenshot:
[image]

I'm using a newly acquired CM4 (I guess latest revision, can't confirm), and a Bigtreetech Manta M4P as carrier.
The M4P get set up with its DIP switches 3 and 4 to ON (like described in the manual [here](https://github.com/bigtreetech/Manta-M4P/raw/master/BIGTREETECH_MANTA_M4P_User_Manual.pdf).

Using alternative cables and ports didn't yield a different result.

### Steps to reproduce the behaviour

Set up carrier as described, attached USB-C to carrier

### Device(s)

Raspberry Pi CM4

### Compute Module IO board.

Bigtreetech Manta M4P V2.1

### RPIBOOT logs

-

### Kernel logs

-

### Device UART logs

-

pelwell, 2023-09-15:
Did you follow the manufacturer's instructions? Have you raised a support ticket with the manufacturer?

Roycinger, 2023-09-15:
> Did you follow the manufacturer's instructions? Have you raised a support ticket with the manufacturer?

Yes, I followed the instructions as per the manual.
I opened a ticket with the manufacturer in parallel

timg236, 2023-09-15:
It would be worth following the troubleshooting guide
https://github.com/raspberrypi/usbboot#troubleshooting

pelwell, 2023-09-15:
Including (from https://bigtreetech.github.io/docs/Software%20Installation.html#cm4-emmc-version):

> in order to avoid problems caused by the insufficient USB power supply of the computer, it is best to use an external 24V power supply to power the motherboard

?

Roycinger, 2023-09-15:
I'll go through the Troubleshooting guide.
I did attach a 24V PSU aswell, just didn't mention that in my initial post

pelwell, 2023-09-15:
Note that the sequencing is important, at least on a CM4IO board - the USB cable must be connected from the host to the CM4IO before the power is applied.

timg236, 2023-09-15:
adding to the comment from @pelwell if the logs are at this stage then it looks like the host has not seen the USB descriptor from the 2711 chip-rom. This could be caused by wrong sequencing, or something else which prevented the ROM from seeing the nRPIBOOT GPIO.

Roycinger, 2023-09-20:
Unfortunately, there's no progress at this point.
Bigtreetech support did answer but didn't seem to understand the problem.
I tried once again by removing the Type-C Power Jumper and attaching direct 24V to the board (as per the documentation on page 10), same result, CM4 doesn't get recognized by the PC while rpi-mass-storage-gadget.bat is running. I also paid attention to the correct sequence as @pelwell described.

nbuchwitz, 2023-09-20:
Any chance you can test the board with rpiboot on a Linux System (eg. live System or rpi)? I've witnessed more than once strange issues where the usb does not work on windows but without any issue on Linux.

Roycinger, 2023-09-20:
Thanks for all your help guys.
The issue seemed to be my Windows machine.
I installed usbboot on a Mac device and the CM4 / Manta was recognized immediately.
I was able to flash the image and the device is available over Wifi now.
Still strange that Windows didn't see the CM4 but I can work with that.

closed 2023-09-20 by Roycinger