raspberrypi/usbboot #188: AvailabilityInternalLegacy.h not found on macOS Sanoma 14.0 (23A344)

Binarytales, 2023-10-09:
### Describe the bug

Following the steps for [building on macOS](https://github.com/raspberrypi/usbboot?tab=readme-ov-file#macos) I get the following error when running `make`:

```sh
$ make
cc -Wall -Wextra -g -o bin2c bin2c.c
In file included from bin2c.c:1:
In file included from /usr/local/include/stdio.h:64:
In file included from /usr/local/include/_stdio.h:69:
In file included from /usr/local/include/Availability.h:168:
/usr/local/include/AvailabilityInternal.h:153:10: fatal error: 'AvailabilityInternalLegacy.h' file not found
#include <AvailabilityInternalLegacy.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [bin2c] Error 1
```

### Steps to reproduce the behaviour

Follow macOS build steps exactly: https://github.com/raspberrypi/usbboot?tab=readme-ov-file#macos

I also ran the _optional_ command to set PKG config path: `export PKG_CONFIG_PATH="$(brew --prefix libusb)/lib/pkgconfig"` which set the path to `/usr/local/opt/libusb/lib/pkgconfig`

---
To fix a previous `make` error:

```sh
$ make                                                   
cc -Wall -Wextra -g -o bin2c bin2c.c
ld: library not found for -lSystem
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [bin2c] Error 1
```
I had to not only update XCode via the AppStore (to version `15.0 (15A240d)`) but also **open and run** XCode to let it install additional packages and libraries. Having only [XCode Command Line Tools](https://download.developer.apple.com/Developer_Tools/Command_Line_Tools_for_Xcode_15/Command_Line_Tools_for_Xcode_15.dmg) installed was not enough.

I also had to run `sudo xcodebuild -license accept` to accept the updated terms and conditions.

--
I am running:
- macOS: macOS Sanoma 14.0 (23A344)
- brew: 4.1.14 
- make: 3.81 (i386-apple-darwin11.3.0)

...on a...
- MacBook Pro 16" 2019
- 2.4 GHz 8-Core Intel Core i9
- 64 GB 2667 MHz DDR4

---

Output of `brew config`
```sh
HOMEBREW_VERSION: 4.1.14
ORIGIN: https://github.com/Homebrew/brew
HEAD: affc4c01aada2c973b63e084e7696e896edf2b7b
Last commit: 8 days ago
Core tap JSON: 08 Oct 22:34 UTC
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 16
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 16-core 64-bit kabylake
Clang: 15.0.0 build 1500
Git: 2.23.0 => /usr/local/bin/git
Curl: 8.1.2 => /usr/bin/curl
macOS: 14.0-x86_64
CLT: 15.0.0.0.1.1694021235
Xcode: 15.0
```
[... 81 lines cut ...]
TargetConditionals.h   com_err.h              float.h                lber.h                 math.h                 pcap-bpf.h             secure                 term.h                 vis.h
Xplugin.h              complex.h              fmtmsg.h               lber_types.h           membership.h           pcap-namedb.h          security               term_entry.h           voucher
__cxxabi_config.h      compression.h          fnmatch.h              ldap.h                 memory.h               pcap.h                 semaphore.h            termcap.h              vproc.h
__wctype.h             copyfile.h             form.h                 ldap_cdefs.h           menu.h                 pexpert                servers                termios.h              wchar.h
_ctermid.h             corpses                fsproperties.h         ldap_features.h        miscfs                 php                    setjmp.h               tgmath.h               wctype.h
_ctype.h               cpio.h                 fstab.h                ldap_schema.h          module.modulemap       png.h                  sgtty.h                tic.h                  webp
_locale.h              crt_externs.h          fts.h                  ldap_utf8.h            monetary.h             pngconf.h              signal.h               tidy                   wordexp.h
_regex.h               ctype.h                ftw.h                  ldif.h                 monitor.h              pnglibconf.h           simd                   tiff.h                 xar
_stdio.h               cups                   get_compat.h           libc.h                 mpool.h                poll.h                 slapi-plugin.h         tiffconf.h             xattr_flags.h
_types                 curl                   gethostuuid.h          libcharset.h           nameser.h              printerdb.h            spawn.h                tiffio.h               xcselect.h
_types.h               curses.h               getopt.h               libexslt               nc_tparm.h             printf.h               sqlite3.h              tiffio.hxx             xlocale
_wctype.h              cxxabi.h               gif_lib.h              libgen.h               ncurses.h              profile.h              sqlite3ext.h           tiffvers.h             xlocale.h
_xlocale.h             db.h                   glob.h                 libkern                ncurses_dll.h          protocols              stab.h                 time.h                 xpc
aio.h                  default_pager          grp.h                  libpng16               ndbm.h                 pthread                standards.h            timeconv.h             zconf.h
aliasdb.h              device                 gssapi                 libproc.h              net                    pthread.h              stddef.h               tk.h                   zdict.h
alloca.h               dirent.h               gssapi.h               libunwind.h            net-snmp               pthread_impl.h         stdint.h               tkDecls.h              zlib.h
apache2                disktab.h              hfs                    libusb-1.0             netdb.h                pthread_spis.h         stdio.h                tkIntXlibDecls.h       zstd.h
apr-1                  dispatch               hid                    libxml2                netinet                pwd.h                  stdlib.h               tkMacOSX.h             zstd_errors.h
ar.h                   dlfcn.h                histedit.h             libxslt                netinet6               python2.7              strhash.h              tkPlatDecls.h
architecture           dns.h                  i386                   limits.h               netkey                 ranlib.h               string.h               ttyent.h
```

### Device(s)

Other

pelwell, 2023-10-09:
bin2c.c is POSIX-compatible c. If that doesn't build on Sonoma then there's not a lot we can do about it.

Binarytales, 2023-10-09:
@pelwell Thanks for your messages. That set me off on a new tangent that lead to me doing the following:

```sh
export CPATH=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/
```

Which fixed the error in my initial bug report. But then I was getting a `ld: library 'System' not found`. A bit more reading and I did:

```sh
export LIBRARY_PATH="$LIBRARY_PATH:/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib"
```

I got 193 warnings about `pointer is missing a nullability type specifier (_Nonnull, _Nullable, or _Null_unspecified)` but these were all from the MacOSX.sdk path so not an issue with `usbboot`.

closed 2023-10-12 by Binarytales