raspberrypi/usbboot #197: Cannot compile on M1 MacBook

cmyk, 2024-01-14:
### Describe the bug

Following the instructions precisely I get this:
```
cmyk@MBP usbboot % make
make: `rpiboot' is up to date.
cmyk@MBP usbboot % sudo ./rpiboot                 
RPIBOOT: build-date Jan 14 2024 version 20221215~105525 20a431c6
Waiting for BCM2835/6/7/2711/2712...
```

### Steps to reproduce the behaviour

Following your instructions on a m1 MacBook with 14.2.1

### Device(s)

Other

pelwell, 2024-01-14:
If you have an rpiboot to execute than compilation has succeeded.

1. What are you trying to do?
2. What is attached?
3. How has it been configured?

timg236, 2024-01-15:
N.B. If you are attempting to run this on Pi5 then you have to run from either the mass-storage-gadget64 or recovery sub-directories. There's no built in "msd.elf" on Pi5 which would explain the missing bootcode5 message.