raspberrypi/usbboot #200: Failed to read correct length, returned 0

CHHE-ledtec, 2024-02-09:
### Describe the bug

RPIBOOT: build-date Feb  9 2024 version 20221215~105525 20a431c6
Waiting for BCM2835/6/7/2711/2712...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Failed to read correct length, returned 0

chhe@pop-os:~/usbboot$ sudo ./rpiboot -d msd
RPIBOOT: build-date Feb  9 2024 version 20221215~105525 20a431c6
Loading: msd/bootcode.bin
Loading: msd/bootcode4.bin
Waiting for BCM2835/6/7/2711/2712...
Loading: msd/bootcode4.bin
Sending bootcode.bin
Failed to read correct length, returned 0

chhe@pop-os:~/usbboot$ sudo ./rpiboot -d recovery
RPIBOOT: build-date Feb  9 2024 version 20221215~105525 20a431c6
Loading: recovery/bootcode4.bin
Waiting for BCM2835/6/7/2711/2712...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to read correct length, returned 0

### Steps to reproduce the behaviour

plugged in new cm4 with waveshare baseboard b

tried it with 2 new units

chhe@pop-os:~/usbboot$ sudo ./rpiboot
RPIBOOT: build-date Feb  9 2024 version 20221215~105525 20a431c6
Waiting for BCM2835/6/7/2711/2712...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Failed to read correct length, returned 0

chhe@pop-os:~/usbboot$ sudo ./rpiboot -d msd
RPIBOOT: build-date Feb  9 2024 version 20221215~105525 20a431c6
Loading: msd/bootcode.bin
Loading: msd/bootcode4.bin
Waiting for BCM2835/6/7/2711/2712...
Loading: msd/bootcode4.bin
Sending bootcode.bin
Failed to read correct length, returned 0

chhe@pop-os:~/usbboot$ sudo ./rpiboot -d recovery
RPIBOOT: build-date Feb  9 2024 version 20221215~105525 20a431c6
Loading: recovery/bootcode4.bin
Waiting for BCM2835/6/7/2711/2712...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to read correct length, returned 0

### Device(s)

Raspberry Pi CM4

### Compute Module IO board.

Waveshare B<se Board B

### Kernel logs

[4565296.966249] usb 1-6.7.2: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[4565296.966257] usb 1-6.7.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[4565296.966260] usb 1-6.7.2: Product: BCM2711 Boot
[4565296.966263] usb 1-6.7.2: Manufacturer: Broadcom
[4565296.966266] usb 1-6.7.2: SerialNumber: f784ddd8
[4565492.086412] usb 1-6.7.2: USB disconnect, device number 119
[4567716.518542] usb 1-6.7.2: new full-speed USB device number 120 using xhci_hcd
[4567716.727869] usb 1-6.7.2: config index 0 descriptor too short (expected 55, got 32)
[4567716.727878] usb 1-6.7.2: config 1 interface 0 altsetting 0 endpoint 0x1 has invalid maxpacket 512, setting to 64
[4567716.727882] usb 1-6.7.2: config 1 interface 0 altsetting 0 endpoint 0x2 has invalid maxpacket 512, setting to 64
[4567716.743847] usb 1-6.7.2: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[4567716.743852] usb 1-6.7.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[4567716.743855] usb 1-6.7.2: Product: BCM2711 Boot
[4567716.743857] usb 1-6.7.2: Manufacturer: Broadcom
[4567716.743860] usb 1-6.7.2: SerialNumber: f784ddd8

timg236, 2024-02-09:
This sounds like a hardware issue with the USB cable / WaveShare board assuming that the host computer is an RPi

Please check the troubleshooting guide 
https://github.com/raspberrypi/usbboot?tab=readme-ov-file#troubleshooting

timg236, 2024-11-05:
Closed due to inactivity

closed 2024-11-05 by timg236