raspberrypi/usbboot #207: macOS: main.c:1:10: fatal error: 'libusb-1.0/libusb.h' file not found

johansmolinski, 2024-03-22:
### Describe the bug

Building for macOS following the instructions fails on finding `libusb-1.0/libusb.h`. Reason is that `main.c` includes `libusb-1.0/libusb.h` where it should include `libusb.h`.

### Steps to reproduce the behaviour

    git clone --recurse-submodules --shallow-submodules --depth=1 https://github.com/raspberrypi/usbboot
    cd usbboot
    brew install libusb
    brew install pkg-config
    export PKG_CONFIG_PATH="$(brew --prefix libusb)/lib/pkgconfig"
    make
    cc -Wall -Wextra -g -o rpiboot main.c bootfiles.c `pkg-config --cflags --libs libusb-1.0` -DGIT_VER="\"8a4b8062\"" -DPKG_VER="\"20221215~105525\""
    main.c:1:10: fatal error: 'libusb-1.0/libusb.h' file not found

The reason being `main.c` is including `libusb-1.0/libusb.h`. Changing to include `libusb.h` works. Here is an analysis of the reason:

    export PKG_CONFIG_PATH="$(brew --prefix libusb)/lib/pkgconfig"
    pkg-config --cflags libusb-1.0
    -I/opt/homebrew/Cellar/libusb/1.0.27/include/libusb-1.0
    ls /opt/homebrew/Cellar/libusb/1.0.27/include/libusb-1.0
    libusb.h

Note that the last line in the output above is `libusb.h` and not `libusb-1.0/libusb.h`.

### Device(s)

Other

nbuchwitz, 2024-03-22:
Possible regression of https://github.com/raspberrypi/usbboot/pull/206/files#diff-a0cb465674c1b01a07d361f25a0ef2b0214b7dfe9412b7777f89add956da10ecR1

Try if it works with the simple 'libusb.h' include.

johansmolinski, 2024-03-22:
Yes. As mentioned, I already tried that as a workaround. Got it working. Just added this ticket to save some time for the next fella. :)

bringert, 2024-03-22:
I see the same problem on Mac with libusb-1.0 installed by Homebrew, and changing the include to 'libusb.h' works around it successfully.

rudelm, 2024-03-23:
It behaves a little bit different on Apple Silicon macs, since Homebrew installs in `/opt/homebrew/opt/`. If Itry the commands from @johansmolinski I'm getting this:

``` bash
export PKG_CONFIG_PATH="$(brew --prefix libusb)/lib/pkgconfig"
pkg-config --cflags libusb-1.0
-I/opt/homebrew/Cellar/libusb/1.0.27/include/libusb-1.0
ls /opt/homebrew/Cellar/libusb/1.0.27/include/libusb-1.0
libusb.h
```

But with the change to the `main.c` its also working on my machine.

RolandasRazma, 2024-03-27:
changing `#include <libusb-1.0/libusb.h>` to `#include <libusb.h>` works on M1 brew

pelwell, 2024-03-27:
Any comments, @tdewey-rpi?

tdewey-rpi, 2024-03-27:
Interesting that on my M1 MBP I required this the other way around. Happy to accept the change.

tdewey-rpi, 2024-11-05:
Resolved many moons ago now. Closing as 'fixed'.

closed 2024-11-05 by tdewey-rpi