raspberrypi/usbboot #210: Raspberry Pi Zero W not recognised by M1 Macbook Air

Hexxeh, 2024-04-01:
### Describe the bug

When connecting a Zero W with no SD card inserted, it is not detected as a USB device by a connected M1 Macbook Air. A Zero 2 W is detected without issue using the same host/cable. I suspect other M1 based devices would see the same issue, and possibly other Apple Silicon machines?

I understand RPi is recommended as a host, I just wanted to report this since I hadn't seen it mentioned in bugs yet and figured it might save someone else some confusion - would it be okay to add a note to the README for this? Happy to make a PR.

### Steps to reproduce the behaviour

Connect a Zero W to an M1 Macbook Air

### Device(s)

Other

sokeman, 2024-08-20:
im having problems with my 2017 i5 macbook air monterey. pi zero 2. it doesnt show anything it doesnt connect it doesnt show anything....    >:(

timg236, 2024-11-05:
Cannot reproduce - closing

closed 2024-11-05 by timg236