raspberrypi/usbboot #225: The secure-boot-example does not tell you how to copy the boot.ug and boot.sig to the eMMC!

torntrousers, 2024-06-24:
### Describe the bug

Following the README here: https://github.com/raspberrypi/usbboot/tree/master/secure-boot-example at the very bottom it says:

Copy the boot.img and boot.sig files from the secure-boot-example stage to the mass storage drive

but it does not tell you how to do that! How do i do that? I can see the eMMC is available at /dev/sda, i could use dd to copy the boot.img but then how to copy the boot.sig???

The README should be updated with the commands to use?
 

### Steps to reproduce the behaviour

.

### Device(s)

Raspberry Pi CM4

### Compute Module IO board.

.

### RPIBOOT logs

.

### Kernel logs

.

### Device UART logs

.

timg236, 2024-06-24:
You would use the mass-storage-gadget as with flashing any other OS image. 

Presumably you have already verified your OS image before enabling secure-boot ?

torntrousers, 2024-06-24:
(also, thanks so much for replying so quickly @timg236 !)

nbuchwitz, 2024-06-24:
Can't use cp here as it copies a file on a Filesystem. Use dd or other tool which. Am write to the raw block dev

Eg. dd if=yourimage.img of=/dev/sda bs=1M conv=fsync

Make sure that sda is the correct path!!

torntrousers, 2024-06-24:
ok, but with secure boot there are two files - boot IMG and boot.sig - so
how can I copy both with the DD command?

nbuchwitz, 2024-06-24:
My bad, you need to copy the files to a Bootable partition (eg vfat) on the emmc for secure boot. See secure boot example Readme file
https://github.com/raspberrypi/usbboot/tree/master/secure-boot-example#loading-bootimg-from-sdemmc

torntrousers, 2024-06-24:
Sorry its still not clear to me.

Its a brand new CM4 so the eMMC is completely empty. Once its mounted with the mass-storage-gadget doe sit need to be formatted? 

Can you help with the exact commands for formatting and/or copying the two boot files?

Many thanks!

timg236, 2024-06-24:
Use mass-storage-gadget to export the devices as block devices. Typically these will appear as /dev/sda 
Format and partition the drive according to your expected layout
Copy the boot.img to the boot file-system.

Installing to EMMC is really outside the scope of this tutorial. It's just running from RAM to demonstrate that an code is loading and running.

torntrousers, 2024-06-24:
Is there some other README somewhere that explains how to do it (i can't find one)? 

There are examples of how to copy a boot.img to the eMMC by using the dd command , but how can that work for the boot.sig too?

I really this this README should explain at least how to deal with coping the boot.sig, or  atleast point to someother README that shows it, surely?

torntrousers, 2024-08-06:
Something like this:

Power OFF and Power ON the CM4, then check what block device it is under lsblk. Then need to build the partitions:
```
sudo umount /dev/sda1 # if necessary

sudo fdisk /dev/sda
```

From here you can create the partitions as follows:

- Type o to create a new MBR partition table
- Create a new boot partition (still inside the fdisk util) with n
-- choose all default values except the last sector where you input +512M (primary, number 1, first sector 2048, last sector +512M)
-- if you get the question "Partition #1contains a vfat signature. Do you want to remove the signature? [Y]es/[N]o:" , Enter Yes
- Create another new partition with n
-- choose all default values (primary, number 2, first sector 1050624, last sector 30535679)
-- Write the modification with w

Reload the partitions with partprobe or unplug and re-plug the CM4.

Next need to format the partitions correctly. Format the first partition as fat and label as boot:
```
sudo mkfs.fat -F 32 -n boot /dev/sda1
```
Then format the second partition as ext4 and label as root:
```
sudo mkfs.ext4 -L root -m 1 0b 4096 /dev/sda2
```
Update the flag of the first partition to "lba" using parted:
```
sudo parted /dev/sda

(parted) set 1 lba on
(parted) q
```
Reload with partprobe or unplug and re-plug the CM4. Reconnect with the rpiboot command:
```
sudo ./rpiboot -d secure-boot-msd
```
If you run lsblk you should see two partitions under sda, and their mounting point (/media/<username>/boot (root)).

torntrousers, 2024-08-06:
is there anything that can get it done simpler?

timg236, 2024-08-12:
The secure-boot-example is there to demonstrate that the code-signing is working and a signed image can be loaded/verified.

If you are creating a new custom OS that requires secure-boot I recommend using the higher level Secure Boot Provisioner tool

https://github.com/raspberrypi/rpi-sb-provisioner

torntrousers, 2024-08-12:
Thank you for the link to the new rpi-sb-provisioner!