raspberrypi/usbboot #231: CM4 is not being detected as as mass storage

drumstick77, 2024-07-11:
### Describe the bug

CM4 EMMC 4GB RAM 16GB storage on Manta M8P V2.0. Not being detected by Windows (win 11), as a mass storage (using RpiBoot)

### Steps to reproduce the behaviour

1.  Toggle switches 4 (USBOTG) and 3 (RPIBOOT) to ON to enter BOOT mode (on Manta M8P)
2. Plus USB C from Manta M8P to Windows 11 laptop
3. Gave the Manta M8P board an external power (PSU) - I can hear the "ding" sound on my laptop
4. Ran RpiBoot which took a while to install
5. No Mass Storage detected.

I've uninstall and re-installed RPIBOOT numerous times (with reboots in every uninstallation), and still a no go. not sure where I've gone wrong

### Device(s)

Raspberry Pi CM4

RavaszEmber, 2024-07-12:
Hello,

I'm having the exact same problem with the exact same hardware. Raspberry PI CM4 on Manta M8 V2.0. XPS 7390 laptop with windows 11. The rpiboot installer took a while (it looped on installing a driver).

When I have my CM4 connected to my laptop via USB C (on the manta board) and run rpiboot I get this:
RPIBOOT: build-date Dec 19 2023 version 20221215~105525 62ce49c0
Waiting for BCM2835/6/7/2711/2712...

It won't progress from here. Please let me know what to do.

pelwell, 2024-07-12:
Contact the manufacturer of the Manta M8.

RavaszEmber, 2024-07-12:
Actually I was able to recognize the eMMC eventually. I used this [post](https://forums.raspberrypi.com/viewtopic.php?t=371123), whose solution was to use a different USB cable.
In my case I went from using this [Anker USB 2.0 C-C cable](https://forums.raspberrypi.com/viewtopic.php?t=371123) to the USB 2.0 C-A (with an A-C adapter for my laptop's usb C port). It worked right away after changing the cable.

It's unclear to me if this is an issue with the cable (they're both USB data cables), the USB C implementation on the Manta M8 V2.0, or the rpiboot software. Since changing the cable did the job it's likely on the Manta M8 V2.0.

pelwell, 2024-07-12:
We can at least say it's an incompatibility between the cable, the carrier, and the use case.

tpups, 2024-09-14:
> Actually I was able to recognize the eMMC eventually. I used this [post](https://forums.raspberrypi.com/viewtopic.php?t=371123), whose solution was to use a different USB cable. In my case I went [...]

Thank you for this comment! I was running into the same issue trying to run rpiboot on my MacBook Pro. Tried two different USB-C to USB-C cables and neither worked. Tried a USB-C to USB-A cable with a USB-A to USB-C adapter on the MBP end, and it worked right away.

Very bizarre, but appreciated.

timg236, 2024-11-05:
This seems to be cable related rather than the software here so closing.

closed 2024-11-05 by timg236