raspberrypi/usbboot #245: rpi-eeprom-config --edit does not commit changes

swess78, 2024-10-11:
### Describe the bug

I've been trying to disable boot from USB so that the rpi does not start in USB mass storage mode. When using the rpi-eeprom-config to edit the config file it does not seem to have the desired effect on BOOT_ORDER= [no combination seems to work or affect the system]

### Steps to reproduce the behaviour

To disable USB boot on your Raspberry Pi using rpi-eeprom-config, you need to modify the EEPROM configuration. Here are the steps:

1. Edit the EEPROM Configuration: Open a terminal on your Raspberry Pi and run the following command to edit the EEPROM configuration:
sudo rpi-eeprom-config --edit

2. Modify the Boot Order: In the editor that opens, look for the BOOT_ORDER parameter. You need to set it to a value that excludes USB boot. For example, setting it to 0xf41 will prioritize SD card boot and exclude USB boot:
BOOT_ORDER=0xf21

3. Save and Exit: Save the changes and exit the editor. The exact method to save and exit depends on the text editor you are using (e.g., Ctrl+X followed by Y and Enter for nano).
Apply the Changes: After saving the changes, reboot your Raspberry Pi to apply the new boot order:
sudo reboot
This should disable USB boot and prioritize booting from the SD card.
 

 

### Device(s)

Raspberry Pi CM4S

timg236, 2024-10-29:
Running "rpi-eeprom-config --edit" won't update the bootloader on a CM4 / CM4-S unless CM4_ENABLE_RPI_EEPROM_UPDATE is set to 1 in /etc/default/rpi-eeprom-update.

The recommended way to modify the bootloader on CM4 / CM4-S  is via rpiboot  - see
https://github.com/raspberrypi/usbboot/tree/master/recovery

This is because CM4/CM4-S cannot loader recovery.bin from the SD/EMMC.

Please could you try the rpiboot approach to see if that works for you.

timg236, 2025-01-28:
Closed due to inactivity

closed 2025-01-28 by timg236