raspberrypi/usbboot #246: New release?

stv0g, 2024-10-22:
### Describe the bug

NixOS is currently only shipping usbboot version 20221215-105525 which is lacking support for the RPi 5.

Could we tag a new release so we can get RPi5 support packaged?

### Steps to reproduce the behaviour

```shell
nix run nixpkgs-unstable#rpiboot
RPIBOOT: build-date Jan  1 1980 version 20221215~105525
Waiting for BCM2835/6/7/2711...
```

### Device(s)

Other

timg236, 2024-10-22:
Here you go

```
commit 82d875522d1a0896e55c90f9e9f47cb4640812c9 (tag: 20240926-102326)
Author: Tim Gover <tim.gover@raspberrypi.com>
Date:   Thu Sep 26 15:32:35 2024 +0100

    win32: Switch to 64-bit Cygwin and update to latest 2711/2712 firmware
    
    * Update for the latest firmware
    * Remove the redundant 32-bit mass-storage gadget
    * Switch to 64-bit Cygwin
```

stv0g, 2024-10-22:
Thanks for the quick action :)

closed 2024-10-22 by stv0g