raspberrypi/usbboot #263: Flash eMMC CM5 Apple

Parnoud, 2024-12-05:
### Describe the bug

Like [This issue](https://github.com/raspberrypi/usbboot/issues/262) you should update the README.

Make doesn't copy file in /usr and if on the latest version of MacOS you try to install it you will get "Operation not permitted". (Yes MacOS security sometime is to much)

like mention on issue 262 you can use -d but you can also modify the makefile to install in /usr/local who is create for this.

Or you can do : `sudo make INSTALL_PREFIX=/usr/local install` this will do the same but in the correct folder for macOS people !

### Steps to reproduce the behaviour

1. Clone the usbboot repository
2. Install libusb (brew install libusb)
3. Install pkg-config (brew install pkg-config)
4. (Optional) Export the PKG_CONFIG_PATH so that it includes the directory enclosing libusb-1.0.pc
5. Build using make
6. Run the binary

```
git clone --recurse-submodules --shallow-submodules --depth=1 https://github.com/raspberrypi/usbboot
cd usbboot
brew install libusb
brew install pkg-config
make
sudo ./rpiboot
```

### Device(s)

Other

### Compute Module IO board.

Compute module 5 IO Board

timg236, 2024-12-05:
I’m happy with readme updates or different makefile default on osx.
We don’t routinely build or test in macOS so normally just merge PRs if they look reasonable

Parnoud, 2024-12-06:
Yes no problem!

closed 2024-12-06 by Parnoud