raspberrypi/usbboot #266: Support Windows Arm64 for rpiboot_setup.exe

ThadHouse, 2024-12-10:
With Windows Arm64 systems becoming more popular, it'd be nice to have an arm64 version of the installer that can install arm64 drivers.

timg236, 2024-12-10:
I think it’s unlikely that we would want to support anymore binary distributions of this.

ThadHouse, 2024-12-10:
I think if the embedded copy of wpi-simple.exe is updated, it will just automatically work. It seems like the latest release of that tool has arm64 support. It shouldn't actually require a specific arm64 installer, at least from what my research is showing.

timg236, 2024-12-10:
It’s not something that we would spend time testing or supporting do I’m closing this as Not Planned.
If patches for the source are required to fix bugs then we normally accept those as is

closed 2024-12-10 by timg236