raspberrypi/usbboot #269: CM5 in a Home Assistant Yellow from Windows

cchesh, 2024-12-28:
### Describe the bug

On running the current rpiboot from Windows cmd I get this:

PS C:\Program Files (x86)\Raspberry Pi> .\rpiboot.exe
RPIBOOT: build-date Nov 13 2024 version 20240422~085300 e3e0fa29
Waiting for BCM2835/6/7/2711/2712...
2712: Directory not specified using default /usr/share/rpiboot/mass-storage-gadget64/
read_file: Failed to read "2712/bootcode5.bin" from "/usr/share/rpiboot/mass-storage-gadget64//bootfiles.bin" - No such file or directory
Failed to open bootcode5.bin

Looks like it's trying to reference a linux file system address/location?

### Steps to reproduce the behaviour

Just run rpiboot on windows with below hardware

### Device(s)

Other

### Compute Module IO board.

Home Assistant Yellow + CM5

cchesh, 2024-12-28:
After reading docs on here (doh!) all I needed to do was add the directory at the command:
.\rpiboot.exe -d .\mass-storage-gadget64

closed 2024-12-28 by cchesh

timg236, 2025-01-06:
The installer also creates a shortcut in the start menu which launches the executable in the correct directory. Unfortunately, the start menu on Windows 11 is less useful these days :(