raspberrypi/usbboot #271: CM4 detected on system but no mass storage

vakeiros, 2025-01-08:
### Describe the bug

CM4 on a BTT MANTA M8p, both switch 3&4 (usb otg, rpi boot) ON, installed rpi boot, cm4 appears on device manager as BCM2711 but I don't get the mass storage unit to install OS. tried everything I could think of... multiple cables, multiple Windows PC, I only didn't tested other CM4 cause I don't have another (this is brand new tho) also didn't test other board (don't have another)test

### Steps to reproduce the behaviour

set switch 3&4 ON on Manta m8p
connect 24v power on M8p
connect USB C to PC
install rpi boot
detects rpi cm4 as bcm2711 on device manger but no mass storage

### Device(s)

Raspberry Pi CM4

### Compute Module IO board.

Manta M8p

nbuchwitz, 2025-01-08:
Did you actually start the rpiboot program? It needs to be started and is not a background service

vakeiros, 2025-01-08:
> Did you actually start the rpiboot program? It needs to be started and is not a background service

Im so stupid... Since it didn't created any shortcut or anything i thought it was just like a driver.... Thanks it worked right away once i actually started the app

closed 2025-01-08 by vakeiros