raspberrypi/usbboot #275: Can't find bootcode files

ofaolin, 2025-01-24:
### Describe the bug

I am trying to run rpiboot on a CM5 for Home assistant yellow, when I run rpiboot according to the instructions for macos, I keep getting this error: No 'bootcode' files found in 'mass-storage-gadget64'

I have tried two different computers, both MacOS, one intel and the other apple silicon.

### Steps to reproduce the behaviour

git clone --recurse-submodules --shallow-submodules --depth=1 https://github.com/raspberrypi/usbboot
cd usbboot
brew install libusb
brew install pkg-config
make
sudo ./rpiboot -d mass-storage-gadget64

### Device(s)

Other

### Compute Module IO board.

CM5

### RPIBOOT logs

RPIBOOT: build-date Jan 24 2025 version 20240422~085300 798ea2ef
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.
No 'bootcode' files found in 'mass-storage-gadget64'
Usage: rpiboot
   or: rpiboot -d [directory]
Boot a Raspberry Pi in device mode either directly into a mass storage device
or provide a set of boot files in a directory from which to boot.  This can
then contain a initramfs to boot through to linux kernel

To flash the default bootloader EEPROM image on Compute Module 4 run
rpiboot -d recovery

For more information about the bootloader EEPROM please see:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-bootloader-configuration

rpiboot                  : Boot the device into mass storage device
rpiboot -d [directory]   : Boot the device using the boot files in 'directory'
Further options:
        -l               : Loop forever
        -o               : Use files from overlay subdirectory if they exist (when using a custom directory)
                           USB Path (1-1.3.2 for example) is shown in verbose mode.
                           (bootcode.bin is always preloaded from the base directory)
        -m delay         : Microseconds delay between checking for new devices (default 500)
        -v               : Verbose
        -s               : Signed using bootsig.bin
        -0/1/2/3/4/5/6   : Only look for CMs attached to USB port number 0-6
        -p [pathname]    : Only look for CM with USB pathname
        -i [serialno]    : Only look for a Raspberry Pi Device with a given serialno
        -j [path]        : Enable output of metadata JSON files in a given directory for BCM2712/2711
        -h               : This help

### Kernel logs

no results

but the system report shows that the device is connected as far as I can tell
BCM2712D0 Boot:

  Product ID:	0x2712
  Vendor ID:	0x0a5c  (Broadcom Corp.)
  Version:	0.00
  Serial Number:	764943f9
  Speed:	Up to 480 Mb/s
  Manufacturer:	Broadcom
  Location ID:	0x14700000 / 8
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

jonhowe, 2025-01-25:
@ofaolin  just had the same issue but on linux, also CM5 + HA yellow.  Copy the firmware/bootfiles.bin to mass-storage-gadget64 and you should be good. There's likely an issue with the latest code here.

erhhung, 2025-01-25:
@jonhowe Thanks for tip. I have the same HA Yellow + CM5, and had initially tried `sudo ./rpiboot -d firmware` since that's where the missing "`bootfiles.bin`" was located, but I believe it needs the rest of the files under "`./mass-storage-gadget/`":

So the working steps on macOS:

```bash
git clone --recurse-submodules --shallow-submodules --depth=1 https://github.com/raspberrypi/usbboot
cd ./usbboot
brew install libusb pkgconf
make
ln -s ../firmware/bootfiles.bin ./mass-storage-gadget/
sudo ./rpiboot -d mass-storage-gadget
```

Which outputs:

```bash
RPIBOOT: build-date Jan 24 2025 version 20240422~085300 798ea2ef
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.
Loading: mass-storage-gadget/bootfiles.bin
Using mass-storage-gadget/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

Second stage boot server
File read: mcb.bin
File read: memsys00.bin
File read: memsys01.bin
File read: memsys02.bin
File read: memsys03.bin
File read: bootmain
Loading: mass-storage-gadget/config.txt
File read: config.txt
Loading: mass-storage-gadget/boot.img
File read: boot.img
Second stage boot server done
```

closed 2025-01-25 by timg236

tschw90, 2025-01-25:
hi, maybe you can help me since my combination (HA yellow & CM5) stucks everytime I prompt 
sudo ./rpiboot -d mass-storage-gadget

saying "Waiting for BCM2835/6/7/2711/2712..."...

using a Mac M1, terminal in rosetta mode or without rosetta makes no difference. tried it while nvme ssd was installed and without nvme ssd.
no errors while installing usbboot.

i also tried to install it with a windows machine. no luck.

hope you can help. very frustrating...

tschw90, 2025-01-25:
> hi, maybe you can help me since my combination (HA yellow & CM5) stucks everytime I prompt sudo ./rpiboot -d mass-storage-gadget
>
> saying "Waiting for BCM2835/6/7/2711/2712..."...
>
> using a Mac M1, [...]

OK ... problem solved. on my HA yellow board, the jumper was not on USB which should have been by default.