raspberrypi/usbboot #278: `mass-storage-gadget64` build of `host-systemd` fails

bryceschober, 2025-01-27:
### Describe the bug

The directions in [`mass-storage-gadget64/README.md#building`](https://github.com/raspberrypi/usbboot/blob/master/mass-storage-gadget64/README.md#building) didn't work for me today:
```shell
git clone --branch mass-storage-gadget64 git@github.com:raspberrypi/buildroot.git
cd buildroot
make raspberrypi64-mass-storage-gadget_defconfig
make
```

At least as of the current revision [`be42d6f4`](https://github.com/raspberrypi/buildroot/commit/be42d6f48aa4e438370ab188ee54fffa499f1ad0) at the head of https://github.com/raspberrypi/buildroot/tree/mass-storage-gadget64, it doesn't build successfully.

I don't know if the problem is a difference in build environment, or what, but it would be helpful if the rudimentary build instructions suggested a host build environment. I know that I am building buildroot's `host-systemd` successfully in another context, of another version.

I got most of the way through that build when it failed during the configure step of `host-systemd 254` with:
```shell
>>> host-systemd 254 Extracting
gzip -d -c /home/bschober/git/rpi-buildroot/dl/systemd/systemd-254.tar.gz | tar --strip-components=1 -C /home/bschober/git/rpi-buildroot/output/build/host-systemd-254   -xf -
>>> host-systemd 254 Patching
>>> host-systemd 254 Configuring
...
Program missing_syscalls.py found: YES (/home/bschober/git/rpi-buildroot/output/host/bin/python3 /home/bschober/git/rpi-buildroot/output/build/host-systemd-254/src/basic/missing_syscalls.py)
Program check-filesystems.sh found: YES (/home/bschober/git/rpi-buildroot/output/build/host-systemd-254/src/basic/check-filesystems.sh)

output/build/host-systemd-254/src/basic/meson.build:234:8: ERROR: Problem encountered: Unknown filesystems defined in kernel headers:

Filesystem found in kernel header but not in filesystems-gperf.gperf: PID_FS_MAGIC

A full log can be found at /home/bschober/git/rpi-buildroot/output/build/host-systemd-254/build/meson-logs/meson-log.txt
make[1]: *** [package/pkg-generic.mk:283: /home/bschober/git/rpi-buildroot/output/build/host-systemd-254/.stamp_configured] Error 1
make: *** [Makefile:82: _all] Error 2
```

I've attached a slightly-redacted log of the buildroot `host-systemd` configure step's output:
[build.log](https://github.com/user-attachments/files/18565582/build.log)

### Steps to reproduce the behaviour

Follow the directions in [`mass-storage-gadget64/README.md#building`](https://github.com/raspberrypi/usbboot/blob/master/mass-storage-gadget64/README.md#building) didn't work for me today:
```shell
git clone --branch mass-storage-gadget64 git@github.com:raspberrypi/buildroot.git
cd buildroot
make raspberrypi64-mass-storage-gadget_defconfig
make
```

At least as of the current revision [`be42d6f4` at the head of `mass-storage-gadget64`](https://github.com/raspberrypi/buildroot/commit/be42d6f48aa4e438370ab188ee54fffa499f1ad0), it doesn't build successfully.

### Device(s)

Other

timg236, 2025-01-28:
Unfortunately, that sounds like a tools/environment issue on your local setup since this configuration doesn't alter core packages like host-systemd.

For reference our build server is Ubuntu 22.04.4 LTS x86 and the command listed above work after a clean build.

bryceschober, 2025-01-28:
Okay, instead of trying to build in one of my internal docker images, I've followed the directions in `buildroot/support/docker/Dockerfile`, getting the image version from `buildroot/.gitlab-ci.yml` as a common public baseline between you and I that should be able to build buildroot:
```shell
$ git clone --branch mass-storage-gadget64 git@github.com:raspberrypi/buildroot.git
$ cd buildroot
$ docker run -it --network=host --volume $(pwd):/home/br-user/buildroot \
  registry.gitlab.com/buildroot.org/buildroot/base:20230207.1123 /bin/bash
br-user@hostname:~$ cd buildroot
br-user@hostname:~/buildroot$ git clean -fdx
br-user@hostname:~/buildroot$ make raspberrypi64-mass-storage-gadget_defconfig
br-user@hostname:~/buildroot$ make
```

And indeed, the build succeeded. I do think that these alternative instructions could be helpful to be in the docs.

pelwell, 2025-01-28:
The instructions appear to be unchanged, except you've added a docker-specific, version-specific, user-specifc line in the middle.

bryceschober, 2025-01-28:
> The instructions appear to be unchanged, except you've added a docker-specific, version-specific, user-specifc line in the middle.

I hear you, but what I've done is use "standard" docker infrastructure, using the same CI docker image as specified (by upstream maintainers) in your buildroot branch, so that the user build environment factor can be mostly taken out of the picture.

Yeah, I get that this extra level of complexity might add more questions from users than those trying to build directly on host unsuccessfully... It was just a suggestion.

I've since discovered that the buildroot docker image is meant to be run via `./utils/docker-run`, so I'll open a PR to suggest that addition to the docs.

PR #279 (2025-01-29): mass-storage-64: Suggest docker build environment
Fixes: #278

Also fixes the path to the expected output image.

Given that I attempted the documented build on my host Ubuntu 22.04.5 LTS system, and the maintainer reported building under the same, it seems reasonable to suggest the built-in upstream docker image as a more host-independent build environment that is known to work.
commit 6cfe7465 (2025-01-29): mass-storage-64: Suggest docker build environment

Fixes: #278

Also fixes the path to the expected output image.

Given that I failed the documented build on my host Ubuntu 22.04.5 LTS
system, and the maintainer reported building under the same, it seems
reasonable to suggest the built-in upstream docker image as a more host-
independent build environment that is known to work.
closed 2025-01-29 by timg236