raspberrypi/usbboot #282: CM4 start4.elf issue?

wills721, 2025-02-20:
### Describe the bug

Trying to use rpiboot to burn Ubuntu Core 20 on a CM4 on MMC.  Trying lots of things but nothing works.  The error at boot it start4.elf is invalid? What am I doing wrong? 

### Steps to reproduce the behaviour

Use rpiboot 

I have tried rpiboot and also rpiboot mass-storage64

### Device(s)

Raspberry Pi CM4

### Compute Module IO board.

CM4 8GB RAM 32GB MMC

JamesH65, 2025-02-20:
You are probably using an out of date Ubuntu image that does not have the required firmware for the latest revision of the CM4. See https://pip.raspberrypi.com/categories/685-app-notes-guides-whitepapers/documents/RP-003476-WP/Updating-Pi-firmware.pdf

wills721, 2025-02-20:
Ah.....I will try this.  But,  I have been trying to download start4.elf but probably (until now I see) its been the wrong one.

The fixup4.dat -- I don't find this anywhere in the usbboot folders though.  I see start4.elf in the mds folder but I don't find a fixup4.dat anywhere?

Thanks for getting back to me so quickly,
Will
@JamesH65

wills721, 2025-02-20:
Def closer now.  I changed the *.elf and fix*dat files in the img and then burned this image.  CM4 boots up almost with Ubuntu Core 20 -- but it will not get past the 'Press enter to configure'.  Not sure if you can help but this is now blocking me.

timg236, 2025-02-21:
usbboot is just a flashing tool for the disk image - please see the Ubuntu forums questions about the OS itself
https://forums.raspberrypi.com/viewforum.php?f=131

closed 2025-02-21 by timg236