raspberrypi/usbboot #295: Add brew installer for OSx

mawildoer, 2025-03-18:
**Is your feature request related to a problem? Please describe.**
It'd be really awesome to install straight with one command :)

**Describe the solution you'd like**
`brew install usbboot` -> Installs `usbboot`

**Describe alternatives you've considered**
Building from source, as the instructions are quite good.

**Additional context**
@gmerlino has done this before: https://github.com/gmerlino/homebrew-rpiboot
This looks to be approximately right and likely only needs a smidge of tweaking to get it up to date after a few years being untouched

Thanks folks!

pelwell, 2025-03-19:
That doesn't look too challenging, but it would require us to create a new repo, `raspberrypi/homebrew-rpiboot`, in order to be able to say `brew tap raspberrypi/rpiboot`, which is a bit tedious. We would also need @gmerlino's licence to use his code.

gmerlino, 2025-03-19:
It's released under MIT license, indeed to have its terms as liberal as possible, so feel free to reuse/evolve as you see fit

pelwell, 2025-03-19:
Thanks - we'll keep that in mind.

mawildoer, 2025-03-19:
> That doesn't look too challenging, but it would require us to create a new repo, `raspberrypi/homebrew-rpiboot`, in order to be able to say `brew tap raspberrypi/rpiboot`, which is a bit tedious. We would also need [@gmerlino](https://github.com/gmerlino)'s licence to use his code.

I can't comment on the friction in the raspberrypi org of getting a new repo up, but if you'd consider it, it'd be a vast upgrade for OSx users!

In order to help with the maintenance, we can support with tap setup and actions to keep it up to date with the main repo. We've got some experience via https://github.com/atopile/homebrew-tap, which we're also currently automating so it's fairly fresh.

If you can confirm the org is happy to accommodate the new repo up and running, we could look into the tap upgrades

gmerlino, 2025-03-21:
@timg236 I'm available, to take on a (light) maintenance role, either on an organization-level (i.e., raspberrypi org) repo, or even the original one (under my name), as you all prefer.

As long as you have PRs and/or suggestions (@mawildoer instance), feel free to send my way: for instance, I'd very much like to accomodate some automations (i.e., actions) in the repo of choice, indeed