raspberrypi/usbboot #297: When passing in a SRC_IMAGE update-pieeprom.sh, it instead uses bootcode.bin from a symlinked EEPROM

bradfa, 2025-03-21:
### Describe the bug

If calling `./update-pieeprom.sh` with  `-i <SRC_IMAGE>` then it appears that during the `sign_firmware()` function that the script will ignore the specified `SRC_IMAGE` path and simply extract and counter-sign the bootcode.bin from an existing hard coded path of `pieeprom.original.bin`, which is symlinked to a specific firmware release within rpi-eeprom submodule.

If the check for `${pieeprom_src}` file existing fails here https://github.com/raspberrypi/usbboot/blob/master/tools/update-pieeprom.sh#L159, then the bootcode.bin is silently not counter-signed and the script still exits with success and produces what appears to be a properly formatted EEPROM image at `${DST_IMAGE}` path.  But inside such an EEPROM will be a bootcode.bin which is not counter-signed.

This seems to have been introduced via dba46ce445427b77dcd4c966c559ccabdc482537

When this silent failure happens it is likely due to not fully or correctly using the rpi-eeprom git submodule, but it is confusing to the user that the rpi-eeprom submodule is required to source an EEPROM input which differs from the one specified by the user via `-i`.

If you use a silent-failure-produced EEPROM to burn the OTP secure boot hash then you will brick your board.  The recovery tool which burns the OTP registers only seems to extract the public key and hash value from the EEPROM image and verify them against each other without ever verifying that bootcode.bin is properly counter-signed to match.

I can understand that there is risk to extracting and counter-signing any given input EEPROM as possibly older Raspi 5 EEPROM bootcode.bin don't properly support secure boot.

I can see 3 possible solutions to avoid bricking like this:

1. Don't hard code `pieeprom_src="pieeprom.original.bin"` but instead use the user's specified input EEPROM image they provided via `-i` to extract the bootcode.bin for counter-signing.
2. Improve the git submodule check at https://github.com/raspberrypi/usbboot/blob/master/tools/update-pieeprom.sh#L204 to actually check for all the needed files and not just the recovery file.
3. When the recovery firmware is burning the secure hash OTP, first verify that the EEPROM which has just been programmed's bootcode.bin actually can be verified by the RSA-2048 public key whose hash is about to be burned into the OTP.

### Steps to reproduce the behaviour

1. `git clone --recurse-submodules https://github.com/raspberrypi/usbboot.git`
2. `cd usbboot/secure-boot-recovery5`
3. `rm pieeprom.original.bin`
4. Copy your RSA 2048 private key into the usbboot/ directory and name it private.pem
5. Add `SIGNED_BOOT=1` to boot.conf
6. `../tools/update-pieeprom.sh -i ../rpi-eeprom/firmware-2712/default/pieeprom-2025-03-10.bin -k ../private.pem -f`
7. Observe in the `hexdump -C pieeprom.bin` output that starting at address 000128c0 there is just a string of `ff`.  This appears to be where the bootcode.bin counter-signature should be.
8. If you feel like bricking a Raspi 5, use the generated pieeprom.bin to burn the OTP secure boot hash.

### Device(s)

Other

bradfa, 2025-03-21:
> pieeprom_src="pieeprom.original.bin" should have been the default and not override the user argument. I've pushed a patch which fixes that.

Thanks!

> I didn't quite follow point (8) as to how the Pi5 could be bricked (permenantly) - as you say, the bootsys signature will be wrong but the OTP key hash should still be correct (or unprogrammed) since it's derived from the signature and public key in the EEPROM image. You should be able to repeat the process (with the linked PR) to sign bootsys correctly.

Sorry, "bricked" probably wasn't the right word to use.  It becomes unbootable without resorting to other methods to recover it.  I attempted to use the usbboot tool to send a variety of what I thought were correctly signed bootcode files, both recoveries and normal, but I was unsuccessful at getting any to load.  This could have been user error on my part in trying to revive the board.

timg236, 2025-03-21:
Yeah, as a bootloader maintainer I regularly make my Pi non-bootable, but rarely permanently so I always worry about how "bricked" a Pi is :)

I think this sequence should make the pi-bootable or at least verify the status 

Edit secure-boot-recovery5 and uncomment this line
`#recovery_metadata=1`

First, try fresh board EEPROM flashing
```
cd secure-boot-recovery5
../tools/update-pieeprom.sh -f -k "${KEY_FILE}"
mkdir -p metadata
sudo ../rpiboot -d . -j metadata -v
```

If that doesn't work add the '-r' flag to countersign recovery.bin
```
cd secure-boot-recovery5
../tools/update-pieeprom.sh -f -r -k "${KEY_FILE}"
mkdir -p metadata
sudo ../rpiboot -d . -j metadata -v
```

If secure-boot is not enabled in secure-boot then bootcode.bin and recovery.bin must NOT be counter-sign. If secure-boot is enabled then the second stage boot (bootcode.bin/recovery.bin) must be signed.

The Raspberry Pi secure-boot provisioner maintains a database of the status of provisioned boards and knows whether or not to countersign recovery.bin
https://github.com/raspberrypi/rpi-sb-provisioner

bradfa, 2025-03-21:
> If that doesn't work add the '-r' flag to countersign recovery.bin
> 
> ```
> cd secure-boot-recovery5
> ../tools/update-pieeprom.sh -f -r -k "${KEY_FILE}"
> mkdir -p metadata
> sudo ../rpiboot -d . -j metadata -v
> ```

Thanks! This did the trick and my board now has a properly counter-signed EEPROM bootcode.bin and it can boot again and I've flashed my desired EEPROM image. :)

Not sure why I couldn't get it to work yesterday, but I understand what's going on now.

timg236, 2025-03-21:
Thanks to trying that. I’ve merged the PR

closed 2025-03-21 by timg236