raspberrypi/usbboot #299: Cannot run update-pieeprom.sh in recovery5

Schuchmann, 2025-03-25:
### Describe the bug

Hi,
if I try to run 
`usbboot/recovery5$ ./update-pieeprom.sh`
I get this error
`../tools/update-pieeprom.sh: 134: SIGN_FIRMWARE: parameter not set`
I am using branch master https://github.com/raspberrypi/usbboot/commit/1d577e1c8186b606acc9804660c2c2ebd704389d
Anyone an idea?

Regards Sven

### Steps to reproduce the behaviour

run `usbboot/recovery5$ ./update-pieeprom.sh`

### Device(s)

Other

### Compute Module IO board.

cm5 IO Board

timg236, 2025-03-25:
I've pushed a fix,  unfortunately, there was some fallout when I added 'set -u' to check for uninitialized variables

Schuchmann, 2025-03-25:
@timg236 Thank you! Works now

closed 2025-03-25 by Schuchmann