raspberrypi/usbboot #306: On Ubuntu rpiboot stops just after 'Successful read 4 bytes'

j75, 2025-05-17:
### Describe the bug

I am trying to flash the eMMC on a Raspberry Pi Compute Module 5 Development Kit.

I followed the instructions but finally it stops with
```
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...
```

According to other issues it should go further and display something like
```
Second stage boot server
File read: mcb.bin
File read: memsys00.bin
File read: memsys01.bin
File read: memsys02.bin
File read: memsys03.bin
File read: bootmain
Loading: mass-storage-gadget/config.txt
File read: config.txt
Loading: mass-storage-gadget/boot.img
File read: boot.img
Second stage boot server done
```

### Steps to reproduce the behaviour

I did

1. `git clone --recurse-submodules --shallow-submodules --depth=1 https://github.com/raspberrypi/usbboot`
2. `cd usbboot/ ; make`
3. `sudo ./rpiboot -d mass-storage-gadget64` then it displayed 
```RPIBOOT: build-date 2025/05/17 pkg-version local 402baf02

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...
```
4. then I connected my computer (running Ubuntu 24.04.2) on the **Power In USB-C** connector
5. now it displays
```
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...
```
...and there it stops!

### Device(s)

Other

### RPIBOOT logs

RPIBOOT: build-date 2025/05/17 pkg-version local 402baf02

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

### Kernel logs

[ 9484.050567] usb 1-1: new high-speed USB device number 11 using xhci_hcd
[ 9484.177077] usb 1-1: New USB device found, idVendor=0a5c, idProduct=2712, bcdDevice= 0.00
[ 9484.177093] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9484.177100] usb 1-1: Product: BCM2712D0 Boot
[ 9484.177105] usb 1-1: Manufacturer: Broadcom
[ 9484.177109] usb 1-1: SerialNumber: 6cd1bdae

timg236, 2025-05-19:
The software installation and commands look correct but it appears that the first full-speed USB transfer is failing. This is sometimes caused by a USB cable issue.

Suggestions

1. Remove any peripherals connected to the CM5IO
2. Remove any hubs between the host computer and the CM5IO or try different USB ports
3. Use a different USB-C cable
4. Try running RPi boot on a Raspberry Pi if you have one available

j75, 2025-05-19:
I run again `rpiboot` but I have added the `v` option - and surprise! everything is completely different!
```
Device located successfully
2711/ position 00000200 size 0
2711/bootcode4.bin position 00000400 size 68696
2711/memsys08.bin position 00011400 size 20492
...
712/font.bin position 0014e400 size 27404
2712/memsys01.bin position 00155200 size 40972
 position 0015f600 size 0
...
 position 00160600 size 0
File 2712/boot.img not found in /home/.../usbboot/mass-storage-gadget64//bootfiles.bin
Loading: /home/.../usbboot/mass-storage-gadget64//boot.img
File size = 31457792 bytes
Received message ReadFile: boot.img
File read: boot.img
libusb_bulk_transfer sent 31457792 bytes; returned 0
Received message Done: boot.img
CMD exit
Second stage boot server done
```

Please note the `File 2712/boot.img not found` message.

The problem is that event after that the CM5 was not displayed as an external storage!

So, I used a Windows system and there I have managed to install the OS on the eMMC.

j75, 2025-05-19:
Concerning the error message, I noticed that the folder **firmware/2711/** has 3 files (2 being symlinks) while **firmware/2712** has only 2 symlinks, is it an error?

pelwell, 2025-05-19:
>  is it an error?

No.

timg236, 2025-05-19:
"Please note the File 2712/boot.img not found message."

With the verbose `-v` option that's expected. It's saying that 2712/boot.img was not found inside the `bootfiles.bin` .tar file. I might bump that to the next level of verbosity

pelwell, 2025-06-06:
Do you not have another Raspberry Pi you can use as an rpiboot host?

Chick92, 2025-06-06:
Hi @pelwell, yes, just tried that and it worked. Good idea! Thanks.

timg236, 2025-10-21:
This seems to be an issue with the Ubuntu rpiboot host USB hardware or OS configuration. Closing as can't reproduce.