raspberrypi/usbboot #318: CM5 Renesas uPD720201K8 mass-storage-gadget crash

TomSievers, 2025-07-08:
### Describe the bug

With a custom CM IO board there is an issue where the mass storage gadget fails to boot and crashes at PCIE initialization with a Renesas uPD720201K8 connected to PCIE. When this chip is disconnect from the board the mass-storage-gadget works normally.

Only occurs on CM5, CM4 seems to not crash even if the Renesas chip is connected to PCIE.

### Steps to reproduce the behaviour

With a CM5 and a Renesas uPD720201K8 connected to PCIE and the nRPIBOOT jumper connected start the rpiboot process with `rpiboot -d mass-storage-gadget64`. rpiboot will finish succesfully, but no usb device will show up.

### Device(s)

Other

### Compute Module IO board.

Custom IO board with a Renesas uPD720201K8 and with SPI EEPROM for the Renesas chip with the usb configuration pre programmed.

### RPIBOOT logs

RPIBOOT: build-date 2025/07/04 pkg-version local e896b5b8

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Boot directory 'mass-storage-gadget64'
Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Device located successfully
Completed file-read 2712/bootcode5.bin in archive mass-storage-gadget64/bootfiles.bin length 73880
Initialised device correctly
Found serial number 3
last_serial -1 serial 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 73880 bytes
libusb_bulk_transfer sent 73880 bytes; returned 0
Successful read 4 bytes 
Waiting for BCM2835/6/7/2711/2712...

Device located successfully
Completed file-read 2712/bootcode5.bin in archive mass-storage-gadget64/bootfiles.bin length 73880
Initialised device correctly
Found serial number 4
last_serial -1 serial 4
Second stage boot server
Received message GetFileSize: mcb.bin
Completed file-read 2712/mcb.bin in archive mass-storage-gadget64/bootfiles.bin length 5460
File size = 5460 bytes
Received message ReadFile: mcb.bin
File read: mcb.bin
libusb_bulk_transfer sent 5460 bytes; returned 0
Received message GetFileSize: memsys00.bin
Completed file-read 2712/memsys00.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys00.bin
File read: memsys00.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys01.bin
Completed file-read 2712/memsys01.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys01.bin
File read: memsys01.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys02.bin
Completed file-read 2712/memsys02.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys02.bin
[... 460 lines cut ...]
[    2.494989] Call trace:
[    2.497438]  readl+0x0/0x18
[    2.500239]  quirk_usb_early_handoff+0x610/0x6fc
[    2.504874]  pci_fixup_device+0xf0/0x1e4
[    2.508811]  pci_bus_add_device+0x28/0x118
[    2.512922]  pci_bus_add_devices+0x58/0x7c
[    2.517031]  pci_bus_add_devices+0x74/0x7c
[    2.521140]  pci_host_probe+0x94/0xac
[    2.524814]  brcm_pcie_probe+0x45c/0x4bc
[    2.528749]  platform_probe+0x68/0xb8
[    2.532423]  really_probe+0x138/0x268
[    2.536096]  __driver_probe_device+0xf4/0x10c
[    2.540467]  driver_probe_device+0x3c/0xf8
[    2.544577]  __device_attach_driver+0x5c/0xd4
[    2.548949]  bus_for_each_drv+0xb0/0xdc
[    2.552797]  __device_attach+0xe8/0x148
[    2.556645]  device_initial_probe+0x14/0x20
[    2.560842]  bus_probe_device+0x3c/0xa4
[    2.564690]  deferred_probe_work_func+0xa4/0xb8
[    2.569236]  process_scheduled_works+0x194/0x244
[    2.573872]  worker_thread+0x144/0x1c8
[    2.577632]  kthread+0xdc/0xec
[    2.580694]  ret_from_fork+0x10/0x20
[    2.584282] Code: f9401bf5 910103ff d50323bf d65f03c0 (b9400000) 
[    2.590399] ---[ end trace 0000000000000000 ]---

pelwell, 2025-07-08:
Please use rpiboot to install an image on at least one of the CM5s using another carrier or in some way disabling the Renesas part, then boot it as normal _with_ the Renesas chip and post the kernel log.

nbuchwitz, 2025-07-08:
Does the renesas controller has a firmware flashed to it's eeprom?