raspberrypi/usbboot #329: Problem to download boot.img to Raspberri PI 5 CM

Heatcliff199, 2025-08-27:
### Describe the bug

We encountered problems when trying to load mass-storage-gadget64/boot.img onto the EMMC memory of the Raspberry PI CM 5.
RPIBOOT: build-date 2025/08/27 pkg-version local 48ec6cb7

```
Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

Second stage boot server
File read: mcb.bin
File read: memsys00.bin
File read: memsys01.bin
File read: memsys02.bin
File read: memsys03.bin
File read: bootmain
Loading: mass-storage-gadget64/config.txt
File read: config.txt
Loading: mass-storage-gadget64/boot.img
File read: boot.img
Failed to write complete file to USB device
```

### Steps to reproduce the behaviour

I turn on the jumper to disable EMMC Boot.
Run command 
`sudo ./rpiboot -d mass-storage-gadget64`
Have some message 
`Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...
`
After I connect the USB cable, I can see that it is charging, but when I last checked, nothing came out. 

### Device(s)

Other

### Compute Module IO board.

[image]

### RPIBOOT logs

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

Second stage boot server
File read: mcb.bin
File read: memsys00.bin
File read: memsys01.bin
File read: memsys02.bin
File read: memsys03.bin
File read: bootmain
Loading: mass-storage-gadget64/config.txt
File read: config.txt
Loading: mass-storage-gadget64/boot.img
File read: boot.img
Failed to write complete file to USB device

### Kernel logs

No errors

Heatcliff199, 2025-08-27:
Have some info about verbose mode 
`➜  usbboot git:(master) ✗ sudo ./rpiboot -v -d mass-storage-gadget64

RPIBOOT: build-date 2025/08/27 pkg-version local 48ec6cb7

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Boot directory 'mass-storage-gadget64'
Loading: mass-storage-gadget64/bootfiles.bin
Using mass-storage-gadget64/bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Device located successfully
Completed file-read 2712/bootcode5.bin in archive mass-storage-gadget64/bootfiles.bin length 73880
Initialised device correctly
Found serial number 3
last_serial -1 serial 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 73880 bytes
libusb_bulk_transfer sent 73880 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

Device located successfully
Completed file-read 2712/bootcode5.bin in archive mass-storage-gadget64/bootfiles.bin length 73880
Initialised device correctly
Found serial number 4
last_serial -1 serial 4
Second stage boot server
Received message GetFileSize: mcb.bin
Completed file-read 2712/mcb.bin in archive mass-storage-gadget64/bootfiles.bin length 5460
File size = 5460 bytes
Received message ReadFile: mcb.bin
File read: mcb.bin
libusb_bulk_transfer sent 5460 bytes; returned 0
Received message GetFileSize: memsys00.bin
Completed file-read 2712/memsys00.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys00.bin
File read: memsys00.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys01.bin
Completed file-read 2712/memsys01.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys01.bin
File read: memsys01.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys02.bin
Completed file-read 2712/memsys02.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys02.bin
File read: memsys02.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys03.bin
Completed file-read 2712/memsys03.bin in archive mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys03.bin
File read: memsys03.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: bootmain
Completed file-read 2712/bootmain in archive mass-storage-gadget64/bootfiles.bin length 575832
File size = 575832 bytes
Received message ReadFile: bootmain
File read: bootmain
libusb_bulk_transfer sent 575832 bytes; returned 0
Received message GetFileSize: config.txt
Loading: mass-storage-gadget64/config.txt
File size = 238 bytes
Received message ReadFile: config.txt
File read: config.txt
libusb_bulk_transfer sent 238 bytes; returned 0
Received message GetFileSize: boot.img
Loading: mass-storage-gadget64/boot.img
File size = 29360640 bytes
Received message ReadFile: boot.img
File read: boot.img
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): no connection to an IOService (code = 0xe00002c0)
libusb_bulk_transfer sent 20643840 bytes; returned -4
Failed to write complete file to USB device
libusb: warning [darwin_close] USBDeviceClose: no connection to an IOService`

pelwell, 2025-08-27:
How are you powering the CM5IO board? You may find it works better with a powered hub between the host computer and the CM5IO board.

Heatcliff199, 2025-08-27:
I connect directly from the macbook usb type-c  to the expansion board type-c , but it's strange that I'm using the same cable as before, and literally 10 days ago I flashed the same system on the same laptop and everything worked. I'll even tell you exactly when I did it: it was August 16.

timg236, 2025-08-27:
If you have a boot-uart connected then it might reveal something but realistically you will either need a Pi4/Pi5 for the rpiboot host computer or a powered USB hub e.g. the Raspberry Pi powered USB hub.

timg236, 2025-08-28:
Are you saying that a previous version of the software works every time? Otherwise, random bulk-transfer issues sound like cable issues problems with MacOS.

Heatcliff199, 2025-09-08:
Hello sorry for to late 
i change cable huy new USB C cable  for transfer data 
and have some errors 
`➜  usbboot git:(master) ✗ sudo rpiboot -v
RPIBOOT: build-date 2025/09/08 pkg-version local 77485809

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Waiting for BCM2835/6/7/2711/2712...

Device located successfully
Directory not specified - trying default /usr/local/share/rpiboot/mass-storage-gadget64/
Completed file-read 2712/bootcode5.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 73880
Initialised device correctly
Found serial number 3
last_serial -1 serial 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 73880 bytes
libusb_bulk_transfer sent 73880 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

Device located successfully
Completed file-read 2712/bootcode5.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 73880
Initialised device correctly
Found serial number 4
last_serial -1 serial 4
Second stage boot server
Received message GetFileSize: mcb.bin
Completed file-read 2712/mcb.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 5460
File size = 5460 bytes
Received message ReadFile: mcb.bin
File read: mcb.bin
libusb_bulk_transfer sent 5460 bytes; returned 0
Received message GetFileSize: memsys00.bin
Completed file-read 2712/memsys00.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys00.bin
File read: memsys00.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys01.bin
Completed file-read 2712/memsys01.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys01.bin
File read: memsys01.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys02.bin
Completed file-read 2712/memsys02.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys02.bin
File read: memsys02.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: memsys03.bin
Completed file-read 2712/memsys03.bin in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 40972
File size = 40972 bytes
Received message ReadFile: memsys03.bin
File read: memsys03.bin
libusb_bulk_transfer sent 40972 bytes; returned 0
Received message GetFileSize: bootmain
Completed file-read 2712/bootmain in archive /usr/local/share/rpiboot/mass-storage-gadget64/bootfiles.bin length 575832
File size = 575832 bytes
Received message ReadFile: bootmain
File read: bootmain
libusb_bulk_transfer sent 575832 bytes; returned 0
Received message GetFileSize: config.txt
Loading: /usr/local/share/rpiboot/mass-storage-gadget64//config.txt
File size = 238 bytes
Received message ReadFile: config.txt
File read: config.txt
libusb_bulk_transfer sent 238 bytes; returned 0
Received message GetFileSize: boot.img
Loading: /usr/local/share/rpiboot/mass-storage-gadget64//boot.img
File size = 29360640 bytes
Second stage boot server done
libusb: warning [darwin_close] USBDeviceClose: no connection to an IOService`

Heatcliff199, 2025-09-08:
Unfortunately, I don't have a Raspberry Pi.

timg236, 2025-10-02:
Latest rpiboot seems to work on MacOS - closing as possible issue with cable or USB ports.
If someone can point to specific USB protocol problem then please re-open.

closed 2025-10-02 by timg236